norbert-walter/esp32-nmea2000-obp60 #22: dev20220309: AWD value displayed instead of AWA

W-Geronius, 2022-03-09:
The windsensor transmits this Apparent Winddirection:
[image]

Webinterface (DATA) of esp correctly displays
[image]

On OBP60 AWA displays as 244 deg in all (one, two, values) formats. 
This is wrong, should be same as above (-116°)

W-Geronius, 2022-03-17:
it's not explicitly defined anywhere that's correct, the majority of commercial products however display 180° +,- or P,S. So it appears to be a convention. In fact I am am not aware any brand displaying 0..360° for AWA.
AWD certainly get's displayed as 0...360° as it is a direction relative to true North

norbert-walter, 2022-03-17:
The problem is the treatment in the formatter. It gets rad as any angular value and converts that to deg. A case distinction would then have to be inserted in the formatter for AWA. However, the formatter does not know what the value is. He only knows that these are angular values. The software design is inadequate at this point.

norbert-walter, 2022-03-20:
I close it.

closed 2022-03-20 by norbert-walter