norbert-walter/esp32-nmea2000-obp60 #26: dev20220309: Time Zones: change of date not calculated properly

W-Geronius, 2022-03-10:
Anchorage: -10h:  10.3. 05:50 UTC displays as **10.3. -5:50 LOT**
Fischi: +12h 10.3. 12.29 UTC displays as **10.3. 24:26 LOT**
watch out for leap years !

norbert-walter, 2022-03-10:
Yes, that is not calculated.

W-Geronius, 2022-03-21:
please check attached files. They hopefully fix LOT date & time calculation within the formatter and the clock page.

As an enhancement I enabled decimal fractions of hours to be processed.
This permits odd timeoffsets like ACWST, Australian Central Western Standardtime (UTC + 8:45)
The ACWST example would require to configure TZ 8.75.
To be added in documentation 

[fixIssue_26.zip](https://github.com/norbert-walter/esp32-nmea2000/files/8314224/fixIssue_26.zip)

norbert-walter, 2022-03-24:
Solved in 20220324

closed 2022-03-24 by norbert-walter