norbert-walter/esp32-nmea2000-obp60 #41: dev20220314 GPSD and GPST displaying LOT ?

W-Geronius, 2022-03-14:
If a timezone is defined, GPSD and GPST display LOT. 
Shouldn't they always be UTC time? Maybe new values local date & local time LOD/LOT should be selectable separately

norbert-walter, 2022-03-16:
When a time zone is defined, GPSD and GPST always show LOT. You also move in the local time. LOT only indicates that the time displayed is not UTC time. The formatter does the conversion to local time. The WebUI is used to determine which time zone you are in. This then applies to the time and date specification. You could also create a special user page that only displays the UTC time. I still wanted to do something like that.

norbert-walter, 2022-03-16:
Thats correct. The data in WebUÍ are always UTC time. The time zone only working for the OBP pages.

closed 2022-03-16 by norbert-walter