norbert-walter/esp32-nmea2000-obp60 #60: OBP60 hardware: LED pairs flickering at random

W-Geronius, 2022-04-15:
At random occasions backlight  LEDs (pairs) were either slightly dimmed or started flickering. Resoldering the right 2 LED + resistor + transistor possibly cured the problem and for a couple of days no further problems were observed.

This morning however, the two leftmost LED started flickering. Cold reboot temporarily solved it. Now the middle two LED are slightly dimmed, the leftmost two minimally flickering, the right two are still perfect.

W-Geronius, 2022-04-17:
are you sure, flickering is totally unaffected by software? First time now I am observing all 6 LEDs flickering - all at the same rate, could be a pattern.

W-Geronius, 2022-04-18:
> First time now I am observing all 6 LEDs flickering - all at the same rate, could be a pattern.

Scratch that, sorry!

Simple solution: looked like too little power and so it was ...
maybe add this to docs!

The display was powered solely via PC USB port:

[image]

btw: 
the PC USB delivers ~5V, just 3.4V are displayed here ?

norbert-walter, 2022-05-19:
Yes, you can not powering the MFD via USB. Any ports supports only 500mA and the voltage loss is to big. Use allways 12V for powering.

closed 2022-05-19 by norbert-walter