norbert-walter/esp32-nmea2000-obp60 #68: E-Paper display fades when exposed to UV, refresh

W-Geronius, 2022-06-04:
Exposed to plain sun the display fades very quickly. As discussed before, an UV protection filter should be attached.

Even the 'full refresh' (seemingly consisting of 2 redraws) doesn't improve display quality - while the first redraw restores normal black the second (and final redraw) even decreases contrast. Attached video demonstrates that effect: refresh occur at 10:55:49 and 11:05:49

Strange enough, the lower part of the screen appears to be less affected by decrease of contrast - it somehow looks lie it 'recovers' from extremely poor readability right after refresh.

To meet github max filesize (10MB) the video is optimized to minimal size and plays at 2x speed:

https://user-images.githubusercontent.com/18160302/172000047-2dcbc7e0-3172-4a29-9523-7eaf7e4ad943.mp4

Original video (~500MB, although not perfect too) is available upon request

The device was exposed to sun, quite hazy though - under clear sun condition the display fades even more quickly

W-Geronius, 2022-06-11:
reminder from threema conversation 
[image]

W-Geronius, 2022-06-24:
reminder to threema conversation:
[image]

promising results using IR filters

norbert-walter, 2022-06-26:
IR radiation ist the problem on the display. The IR radiation heating up the black pixel and reducing the viscosity of gel layer between the glases. To resolf the problem we can using sunshine protection foil with UV and IR blocker. Several manufacturer has helpfully products. Products from 3M are the best.

closed 2022-06-26 by norbert-walter