norbert-walter/esp32-nmea2000-obp60 #72: XTE formatting and default formatting of negative numbers

W-Geronius, 2022-07-22:
1.) OBP60Formatter.cpp lacks specific XTE handling  (like value->getFormat() == "formatXTE"). 
Therefore no unit is displayed for XTE values and formatting is left to default processing (lines 684ff)

2.) XTE = 0 is displayed as **-0.00**

3.) default processing formats negative values with 2 decimals, causing an overflow on display (e.g. in bottom fields of PageFourValues2)

W-Geronius, 2022-07-22:
as a quick & dirty fix, default formatting could be modified:

> {
            if(value->value >= 100){
            snprintf(buffer,bsize,"%3.0f",value->value);
            } else if (value->value >= 10){
            snprintf(buffer,bsize,"%3.1f",value->value);
            } else if (value->value > 0){
            snprintf(buffer,bsize,"%3.2f",value->value);
            } else if (value->value > -10){
            snprintf(buffer,bsize,"%3.1f",value->value);
            } else {
            snprintf(buffer,bsize,"%3.0f",value->value);
            }

W-Geronius, 2022-08-25:
possibly the dirty fix leads to this:
[image]

[image]

odd XTE value !

to be investigated

norbert-walter, 2024-07-26:
Good idea

closed 2024-11-02 by norbert-walter