norbert-walter/esp32-nmea2000-obp60 #80: OBP60 V2: display clearly fades when exposed to sunlight

W-Geronius, 2024-06-05:
possible reasons:
display hardware, IR and/or UV filter
the newly implemented refresh type (quick refresh) is not suited to completely recover clarity
testing by halogen lamp does not uncover the weaknesses as real sun (on water?) 
any findings and tests for improved filters / setups / refresh techniques may be documented here

Liese2007, 2024-06-09:
I have same effects. 

[image]

[image]

15°C outside under Cockpit tent.
OBP60 running since 1,5 days.
Will check if I can find OBP60 inner  temperature value.

Liese2007, 2024-07-24:
Actually in Kopenhagen, with morning sun, it looks even worth. Cannot get the inner temperature. No reaction by swiping.
[image]

W-Geronius, 2024-07-24:
that exactly corresponds to my device. At a certain point all buttons become irresponsive - no swiping, no keylock, no light switching. It seems like the thread handling button events just stops working - GPS however continues functioning, time and even WifI work fine

 Even after cooling down over night, the fading effect persists - after reboot the screen looks okay - however, just for a few minutes!

BEWARE: while experimenting with the half dead device, I tried a factory reset, resulting the device to end up in a even worse state: startscreen not set, no AP started, no connection to WiFi. So a manual complete flash via USB is required ... be prepared for that!

norbert-walter, 2024-07-26:
The error is understandable and reproducible. Despite IR and UV filters, the e-paper display is affected by direct and sometimes indirect sunlight. The problem lies with the e-paper display itself. The display manufacturer has added a new display type and the old display type is no longer produced. The control chip and the wavetables have also been changed and optimized for speed. No problems were found in laboratory tests with a 200W halogen lamp without a UV filter as a sun simulator.

With partial display refreshes, there is an increasing loss of contrast. A full refresh initially corrects the loss of contrast. As time goes on, even the full refresh cannot completely correct the loss of contrast. If the OBP60 is switched off and removed from sunlight, the display recovers after a while.

It is currently not entirely clear what causes the error or which wavelength range of light is responsible for it. It would be possible to solve the problem with customized filters or wavetables. To do this, the manufacturer of the display must be contacted to find a solution.

norbert-walter, 2024-07-26:
The higher contrast loss in the upper left area is due to temperature. The power loss of the ESP32-S3 of around 2W causes a temperature increase in that area. The problem only becomes clearly visible when the temperature in the housing exceeds 50°C.

The temperature increase can be reduced by reducing the CPU's power. Whether this is sufficient still needs to be tested. A heat sink that distributes the temperature over a large area would also be conceivable.

Liese2007, 2024-07-26:
Thanks,
I took 3 measurements before start , after 1min20s and after 3min. The temperature has no or only minor influence?
I measured on the display with a Testo 830 T1 laser measurement.
Indirect sunlight. Cloudy day.[image]

[image]

[image]

I was wondering, that the Open Boat Projects Logo has always been cutting sharp with good contrast, while the WiFi QR Code has already noticeable loss of contrast....

Greets

norbert-walter, 2024-07-26:
Was the display for a longer time switched off? The display need this off time for repair the display function.

Liese2007, 2024-07-26:
> You need more time for heating up the display. App. 20 min or more. Was the display for a longer time switched off? The display need this off time for repair the display function. 

It was switched off the whole night.
And why is the huge difference in contrast between logo and WiFi QR. No idea....

norbert-walter, 2024-07-26:
This is an indication that the problem has something to do with the display control. The wavetables describe the type of display control. I hope that the problem can be solved with adapted wavetables. But I need the manufacturer's help for this. The manufacturer has very special wavetables for different applications.

Liese2007, 2024-07-29:
Another hint, that the display control is related imo is, that actually with a static OBP Logo there is no loss of contrast at all in the sun.

Temperature not measured yet.

norbert-walter, 2024-08-09:
After some consideration and testing, I believe that I am not controlling the display correctly via software. I suspect why this might be, as the errors only occur during a partial refresh. The timing of the commands seems to be the problem. I will try to change the software accordingly.

norbert-walter, 2024-10-03:
I reduced the fading with new power modes and new refresh modes for OBP60 with firmware dev20240920. But the effect is only reduced and not completely fixed. The problem is deeper in display hardware and wave tables.