norbert-walter/esp32-nmea2000-obp60 #81: OBP60 V2: sporadic reboots and rare partial freezing of display

W-Geronius, 2024-06-05:
Reboots and system hangs occur sporadically. This can also be observed with a completely original setup (out of the box, standalone)
Reboots are frequent, several within one day. Occasionally (1 to 2 times per day) the system crashes partially: the display freezes, will not updated. Buttons and buzzer continue to work (wipe, keylock - not sure about illum). The flashing LED inside the housing suggests that the GPS module is still working as well.

To facilitate joint troubleshooting, all observations and tests should be collected here.

W-Geronius, 2024-06-18:
it appears that reboots occur more when frequently, when AP is active - could also be observed on OBP60 V1

norbert-walter, 2024-07-26:
The cause of the firmware crashes is very complex, as 2 CPUs are used with multiple tasks. The causes have not been located at the moment. There are many indications that the problems are related to data traffic over WiFi. As a result, some tasks crash and the device is only partially functional. Firmware recovery is also affected. In some cases, the flash memory is corrupted and parts of the settings are lost. Firmware recovery is then only possible via USB-C.

norbert-walter, 2024-08-07:
This function is already implemented in the firmware. Under Config - System, a time period for the access point can be defined (stopApTime). The value is given in seconds and defines how long the access point should be active after a start. Test the function and check whether the error is then reduced. Please let us know the result.

Please refer the documentation: https://obp60-v2-docu.readthedocs.io/de/latest/usermanual/configuration.html#config-system

Unoxa, 2024-08-08:
Thanks, I will try that and report back. Won’t be on the boat for two weeks
now but after that.

norbert-walter, 2024-08-09:
I turned off the WiFi after a while. The crashes stopped. Apparently this is a temporary solution.

norbert-walter, 2024-10-03:
Fixed in dev20240920

closed 2024-10-03 by norbert-walter