norbert-walter/esp32-nmea2000-obp60 #82: dev20240723: Configuration: Factory Reset incomplete

W-Geronius, 2024-07-24:
After conducting a factory reset via the device's configuration page, the user is locked out of the system: neither does it create the default access point, nor does it link to a previously defined WiFi. The startupscreen doesnt get resetted either, it still displays the previously configured startpage. Apparently, some extra memory locations specific to the OB60's modifications of the original software (esp32-n2K) require extra treatment by the reset routine.

A complete erase & flash via USB is required in this state

norbert-walter, 2024-07-26:
We have to fix the problem in the project esp32n2k from wellenvogel. That is a part of his project.

W-Geronius, 2024-10-03:
? fixed in dev20240920

norbert-walter, 2024-10-03:
Fixed in dev20240920

closed 2024-10-03 by norbert-walter