norbert-walter/esp32-nmea2000-obp60 #83: Document improvement: additional Info to be added to chapter 'webflashtool'

W-Geronius, 2024-07-24:
**Trennen Sie das OBP60 von der Stromversorgung.** Öffnen Sie die hintere Gehäuseabdeckung und stellen Sie mit der Drahtbrücke eine Verbindung von GND (CN2) zum Pin 27 (ESP32-S3) her. **Stellen Sie die Stromversorgung wieder her - das OBP60 startet nun nicht wie gewohnt und gibt auch keinen Beeplaut von sich. Mit weiterhin bestehender Brücke verbinden Sie dann** das OBP60 mit dem PC über das USB-Verbindungskabel. Sobald sie USB-Schnittstelle erkannt wird, erfolgt eine Tonausgabe auf dem PC.

norbert-walter, 2024-07-26:
In the meantime, I have changed the documentation in some places, but not in the suggested wording.

norbert-walter, 2024-08-19:
Is implemented

closed 2024-08-19 by norbert-walter