norbert-walter/esp32-nmea2000-obp60 #84: Updating firmware fails due to 'magic byte' error for all xxxx-all.bin

W-Geronius, 2024-07-24:
[image]

xxxxx-all.bin never get installed due to this error, irrespective of their origin (github or gitpod)
xxxxx-update.bin install without problems.

Don't try to 'emulate' an -all update by factory reset! See previous issue!

norbert-walter, 2024-07-26:
I'm not sure if the problems are related to a corrupt flash memory. See Issue #81

Liese2007, 2024-08-17:
I get the same error with the test file.
[image]

norbert-walter, 2024-08-18:
Please unzip the file and use the bin file.

W-Geronius, 2024-10-19:
opened issue wellenvogel/esp32-nmea2000/issues/89

W-Geronius, 2024-10-19:
Andress responded: https://github.com/wellenvogel/esp32-nmea2000/issues/89#issuecomment-2424080702

this should be included in OBP60 documentation.