norbert-walter/esp32-nmea2000-obp60 #85: Housindg deformed after sun exposure

W-Geronius, 2024-08-05:
After exposure to NL sun for 1 day, the lower part of the housing is deformed. This means that it is no longer watertight.

[image]
[image]

norbert-walter, 2024-08-09:
I also noticed this problem on one of my displays but not so extrem. However, it only occurs when the display is not mounted. If the display is mounted on a flat surface, the deformation should disappear. It may help to put the part in the oven at 80...90°C with the back on a glass plate. After cooling, it should be flat again.

Currently, PLA was chosen as the material, which heats up to around 90°C when exposed to direct sunlight. Other materials would certainly be more suitable, such as PETG.

norbert-walter, 2024-10-03:
I want to use ABS for housing. I think then is it solved.

TobiasE-github, 2025-04-12:
My hosing also deformed after 2 weeks in the lake Constance sun. This is like liked to the asymmetrical pressure distribution in the seal, as only 2 screws are used

[image]
[image]