norbert-walter/esp32-nmea2000-obp60 #87: LED sporadically flash in a different color

W-Geronius, 2024-08-05:
If the display is set to red (others haven't been tested so far), flashes of a different color will appear sporadically - blue mostly, but sometimes green flashes have also been observed 

https://github.com/user-attachments/assets/2480a0d9-ef9d-470d-b298-47ac95a743de

https://github.com/user-attachments/assets/dcf23e28-f261-4d80-91df-a34c649bc251

norbert-walter, 2024-08-09:
The RGB LEDs are controlled serially at 1 MHz. Due to the high number of tasks running in the firmware, data transmission errors occasionally occur. The RGB LEDs are controlled cyclically, so the errors only occur for a short time.

As a solution, the cyclic control could be replaced by an event control when the state of the LEDs changes. This should stop the flickering.

W-Geronius, 2024-10-03:
seems to be gone in dev20240920

norbert-walter, 2024-10-03:
Fixed in dev20240920

closed 2024-10-03 by norbert-walter