norbert-walter/esp32-nmea2000-obp60 #88: dev20240920: Flash LED Mode 'GPS fix lost' stopped working

W-Geronius, 2024-10-03:
Flash LED Mode 'GPS fix lost' stopped working. 

setup: 
internal GPS deactivated, GPS data fed via WiFi (AvNav/SignalK)
stopping GPS doesn't trigger LED anymore

norbert-walter, 2024-10-03:
The LED indicator is only working for internal hardware GPS and not working for external GPS data. This is not a bug.

W-Geronius, 2024-10-03:
For OBP60 V1 this setting causes the LED to flash, when no GPS data are received. It serves extremely well as indicator for a broken wifi-connection - not an uncommon event in a raspberry pi (openplotter) environment. In particular with 'hold values' set to 'on' a dropped wifi-connection will pass completely unnoticed.

W-Geronius, 2024-10-07:
I will further explore this issue, as under certain (rare?) circumstances, the LED in fact flashes and message 'no GPS data' is shown even if internal GPS is set to 'off'.

W-Geronius, 2024-10-10:
From what I found, LED flash mode 'GPS fix lost' seems to be solely triggered by the GPSD and GPST.

[image]

The LED will never flash whenever GPSD and GPST are present, irrespective of GPS fix

W-Geronius, 2024-11-03:
fixed by latest commit

closed 2024-11-03 by W-Geronius