norbert-walter/esp32-nmea2000-obp60 #89: dev20240920: 'save/restart' doesn't reset ILUM (backlight)

W-Geronius, 2024-10-03:
other than in OBP60 V1, restarting the device via configuration will not change backlight illumination anymore. However, this may be considered as a feature: 
IRL 'keylock' sometimes needs to be released via configuration/restart - esp. in night conditions, preserving the ILUM state is a good thing!

norbert-walter, 2024-10-07:
It is not a feature. It is unintentionally but the function is not bad especially at night during a reboot.

W-Geronius, 2024-10-10:
There is a difference between 'restart' via web interface and 'reboot'.

see definition of ESP.restart() in espressiv library
* Peripherals (except for WiFi, BT, UART0, SPI1, and legacy timers) are not reset.

Unintentional reboots (like when connecting/disconnecting a terminal to USB) will reset ILUM.
Are there any other settings that should in fact be reset on restart?