norbert-walter/esp32-nmea2000-obp60 #95: dev20240920: GPSD and GPST incorrectly set

W-Geronius, 2024-10-10:
referring to #88, date and time fields are populated with odd values, when:

1) internal GPS is defined, but RTC is set to off: right after restart
2) both internal GPS and RTC are defined: after a while if GPS doesn't obtain a fix

[image]

The OBP60 device displays 08:32:06 05.06.1905 LOT for this dataset

Question: how is the RTC supposed to be updated with correct date & time?

norbert-walter, 2024-10-17:
The RTC time is automatically set every 5 minutes if a GPS time is available. Manual setting of the RTC is not currently provided in the firmware. The reason for this behavior may be that the RTC did not have a valid time at the start time and is displaying the time that is randomly set in the RTC. The time is only transmitted and stored in the RTC when a valid GPS signal is available. Another possible reason is that the backup battery is empty.

We need to investigate the behavior more closely.

W-Geronius, 2024-10-23:
You name it: **_... when a valid GPS signal is available ..._**
That's the problem allover: I can't find a (generally avaible) check for a GPS fix - at present this kind of check is performed by inspecting GPSD and GPST. This was ok as long as no RTC was available. In V2 however this causes all sorts of faulty behaviour, on clock page as well as with 'GPS signal lost' alert. Possibly others as well

W-Geronius, 2024-10-23:
Having the HDOP threshold configurable is a very good idea. Within my testing environment @home, the system was accepting a GPS fix with HDOP>25, the position was arbitrarily jumping within a radius of several hundreds of meters.

norbert-walter, 2026-02-06:
Solved with dev20260206