norbert-walter/esp32-nmea2000-obp60 #97: Confirm option: use sog instead of stw

TobiasE-github, 2024-10-12:
It would be nice to have a config option to use sog instead of stw on all display pages

norbert-walter, 2024-10-17:
Okay, that's understandable. The changes have to be made in the NMEA2000 gateway. Please open an issue using the following link. I can't make a pull request in this project because we want to keep the projects independent of each other.

https://github.com/wellenvogel/esp32-nmea2000

TobiasE-github, 2024-10-17:
Alright, I'll open an issue over there. 

I really like the wind rose display, so another way of solving this could also be to make some fields user-defined, instead of the hard coding STW and DBK in line 408 of Page wind rose.cpp

closed 2025-01-18 by TobiasE-github