norbert-walter/esp32-nmea2000-obp60 #113: PageClock: Minute hand

thooge, 2024-11-13:
The minute hand of the clock jumps back and forth
This appears if minute is changing.
Hand goes one minute further, then jumps back and then again forward.

norbert-walter, 2025-01-04:
In wich firmware version is the problem? I can not see the same problem on my device.

thooge, 2025-01-05:
Seems to be concerned RTC drifting away and no current gps fix.
I have a rework of rtc/gps date/time behaviour in my pipeline (almost done),
perhaps that fixes the issue.

norbert-walter, 2025-01-26:
You using two GPS sources at the same time? Very often is this the problem. Both times has a little difference.

thooge, 2025-01-26:
Im using OBP60 as standalone device.
No GPS fix.

norbert-walter, 2025-02-03:
I can not reproduce the error on my system.

thooge, 2025-02-04:
I have code improvement for the clock-page and assume the behavoiur 
is inside the gateway / gps code.
I will go into deeper inspection and report here.

thooge, 2025-02-05:
I logged the data sent to N2K-Bus from OBP60.
Seems there are two sources: Local Crystal clock is generated by OBP60Sensor task.
GPS seems to generated by the gateway itself.
`
  source=32, date=2025-02-05, secs=59584.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59212.0, ts=GPS
  source=32, date=2025-02-05, secs=59584.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59585.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59213.0, ts=GPS
  source=32, date=2025-02-05, secs=59585.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59214.0, ts=GPS
  source=32, date=2025-02-05, secs=59586.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59586.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59215.0, ts=GPS
  source=32, date=2025-02-05, secs=59587.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59587.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59216.0, ts=GPS
  source=32, date=2025-02-05, secs=59588.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59588.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59217.0, ts=GPS
  source=32, date=2025-02-05, secs=59589.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59589.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59218.0, ts=GPS
  source=32, date=2025-02-05, secs=59590.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59590.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59219.0, ts=GPS
  source=32, date=2025-02-05, secs=59591.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59591.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59220.0, ts=GPS
  source=32, date=2025-02-05, secs=59592.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59592.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59221.0, ts=GPS
  source=32, date=2025-02-05, secs=59593.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59593.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59222.0, ts=GPS
  source=32, date=2025-02-05, secs=59594.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59594.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59223.0, ts=GPS
  source=32, date=2025-02-05, secs=59595.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59595.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59224.0, ts=GPS
  source=32, date=2025-02-05, secs=59596.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59596.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59225.0, ts=GPS
  source=32, date=2025-02-05, secs=59597.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59598.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59226.0, ts=GPS
  source=32, date=2025-02-05, secs=59598.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59599.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59227.0, ts=GPS
  source=32, date=2025-02-05, secs=59599.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59228.0, ts=GPS
  source=32, date=2025-02-05, secs=59600.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59600.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59229.0, ts=GPS
  source=32, date=2025-02-05, secs=59601.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59601.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59230.0, ts=GPS
  source=32, date=2025-02-05, secs=59602.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59602.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59231.0, ts=GPS
  source=32, date=2025-02-05, secs=59603.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59603.0, ts=Local Crystal clock
  source=32, date=2025-02-05, secs=59232.0, ts=GPS

`