norbert-walter/esp32-nmea2000-obp60 #168: Clock Page: RTC/GPS Mode

W-Geronius, 2025-05-08:
Some releases ago the clock page has been modified to display either RTC or GPS time. Default is RTC
This is a nice feature as long as a RTC is present and operating. 

However for devices lacking a RTC (like OBP40) or systems with RTC disabled via configuration, it's a nuisance:
at least one keypress is required after every reboot of the device to display valid time

Rather than introducing another configuration option, I strongly suggest to use GPS as default - GPS is more likely to be available than RTC

thooge, 2025-06-12:
For OBP40 the default time source is already GPS.
Also  a special option "iRTC" is introduced for OBP40: It will try to find a NTP-Server and sets the ESP32 internal RTC to that value.
Note: The OBP40 has no internal GPS

W-Geronius, 2025-06-12:
My OBP40  defaults to RTC, for whatever reason - for OBP60 clock start display should not be RTC too if RTC is not present (disabled)

W-Geronius, 2025-07-06:
In release dev20250703 (binaries provided in SF by Norbert), the OBP40 still defaults to RTC. Same for OBP60 (with RTC disabled). I am not sure if this issue had beed adressed in any of the recent modifications.

thooge, 2025-07-07:
Modifications coming soon. On my list.

norbert-walter, 2025-07-08:
I noticed that it constantly switches between RTC and GPS as long as no GPS time is found. The time then randomly switches back and forth on the display. If a valid time is found via GPS and synchronized with the RTC, both clocks are synchronized. Apparently, there's still an error in correctly selecting and displaying the valid time. I only noticed the problem because my backup battery was dead, and the time in the RTC wasn't saved and continued to run correctly.

thooge, 2025-07-08:
Investigating generic date and time behaviour. 
Seems that boatvalues GPSD, GPST and HDOP considered valid 
even if standalone device with no GPS and no network connection.

norbert-walter, 2026-02-06:
Solved with dev20260206