norbert-walter/esp32-nmea2000-obp60 #193: Proposal: disable mode X in PageWind.cpp

TobiasE-github, 2025-08-03:
PageWind is really useful in mode L and N,  mode X is not so useful (yet). Skipping Mode x in daily use is annoying.  I suggest commenting out lines 250 an 251.

TobiasE-github, 2025-09-27:
Mode N ("Normal") draws a boat-like shape, and displays the wind direction for the full range of +- 180 degrees.
Mode L ("Lens")  is similar, but magnifies a range of wind angles.
Mode X ("eXample") displays a circle and the wind direction, and has buttons to make the circle bigger and smaller. 

I find myself using mode N and L regularly. Switching modes works by cycling through them, so to get from L to N you always go through X. 
 
I (personally) don't think that mode x adds value to the other modes, so i suggest just disabling it in the selection code.

thooge, 2025-09-27:
That mode is a placeholder for the example zoom-in / zoom-out code.
Lens-symbols in footer.
But i agree that code should be disabled. Put it onto my todo list.

TobiasE-github, 2025-10-03:
Pull Request #206 disables mode X